NVIDIA/jetson-gpio #98: GPIO on non-development carrier boards

dahliarobotics, 2023-11-13:
I am using a Jetson Xavier NX with on of the recommended carrier board from avermedia: [EN715](https://www.avermedia.com/professional/product-detail/EN715#download).
How can I utilized the GPIO header of such a board? Is it possible with this repo? Any changes needed to make it work? Or is there any more general package for non-development boards?

anhmiuhv, 2023-11-14:
No it is not possible. You can read more about GPIO programming on Linux in general here

https://www.ics.com/blog/gpio-programming-exploring-libgpiod-library

jmachuca77, 2023-11-27:
You would need to modify the gpio_pin_data.py file with the pins your carrier board has. Fair warning the pin numbering on the Jetsons is a nightmare. So figuring out the values that go in that table is not very straight forward. You will need to have access to the schematics of your board and these files:
[Pinmux spreadsheet](https://developer.nvidia.com/downloads/jetson-orin-nx-and-orin-nano-series-pinmux-config-template) 
[Identifying the GPIO Number](https://docs.nvidia.com/jetson/archives/r35.4.1/DeveloperGuide/text/HR/JetsonModuleAdaptationAndBringUp/JetsonOrinNxNanoSeries.html#identifying-the-gpio-number)