NVIDIA/jetson-gpio #99: warn_if_not_carrier_board error check

Wu-Fisher, 2023-11-14:
I am using the Jetson Orin NX developer kit-16GB. 
When I imported jetson.gpio, the terminal showed message of
"WARNING: Carrier board is not from a Jetson Developer Kit.
WARNNIG: Jetson.GPIO library has not been verified with this carrier board,
WARNING: and in fact is unlikely to work correctly."
So I went to check the source code. I got "
        elif matches(compats_jetson_orins_nx):
            warn_if_not_carrier_board('3509', '3768')
            return JETSON_ORIN_NX" in "gpio_pin_data.py:585"
That would check the file "/proc/device-tree/chosen/ids" to see if 3768 is the prefix.
But my own file is only 3767.
I am a green hand of such devices, so there might be some error description.

anhmiuhv, 2023-11-14:
3767 is the module not the carrier board. Is your target official NVIDIA Jetson Orin NX? 

You could also this repo anyway to see if it works or not if it is official NVIDIA Jetson Orin NX.

Wu-Fisher, 2023-11-18:
> 3767 is the module not the carrier board. Is your target official NVIDIA Jetson Orin 
[image]

It should be official.right?

anhmiuhv, 2023-11-18:
Yeah you can try to ignore the warning and run it

Wu-Fisher, 2023-11-18:
yep, that is what I had done.

anhmiuhv, 2024-01-05:
If you have no more problem I will close this.

closed 2024-01-05 by anhmiuhv