NVIDIA/jetson-gpio #103: wait_for_edge doesn't timeout/respond on Orin Nano

Romihi, 2024-01-04:
Hello.
I am trying to use GPIO pins on Jetson Orin Nano for my project.
It seems wait_for_edge doesn't timeout or respond (It works on my Jetson Nano).
I checked input and output to function, thought. 

**To Reproduce**
When running "samples/button_event.py", it does not time out .
even adding timeout millisecond such as:
            GPIO.wait_for_edge(but_pin, GPIO.FALLING,500)
 in line 48.

The command line shows like below:
~~~
Starting demo now! Press CTRL+C to exit
Waiting for button event
^CTraceback (most recent call last):
  File "samples/button_event.py", line 60, in <module>
    main()
  File "samples/button_event.py", line 49, in main
    GPIO.wait_for_edge(but_pin, GPIO.FALLING,500)
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 542, in wait_for_edge
    result = event.blocking_wait_for_edge(ch_info.chip_fd, ch_info.gpio_chip, channel, request, bouncetime, timeout)
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio_event.py", line 465, in blocking_wait_for_edge
    ret = select.select([request.fd], [], [], timeout)
KeyboardInterrupt

anhmiuhv, 2024-01-05:
you have to use
GPIO.wait_for_edge(but_pin, GPIO.FALLING,timeout=500)

Also the timeout is in seconds, not miliseconds.

Romihi, 2024-01-06:
@anhmiuhv 
Thank you for replying!
There is still an issue for looping "wait_for_edge" in the sample code. 
Should I use "add_event_detect" and callback instead to check pins' HIGH/LOW several times?
(As in "https://stackoverflow.com/questions/38125047/raspberry-pi-runtimeerror-conflicting-edge-detection-already-enabled-for-this-g")

I got now the following "RuntimeError: Conflicting edge detection event already exists for this GPIO channel".
~~~~ 
Starting demo now! Press CTRL+C to exit
Waiting for button event
Button Pressed!
Waiting for button event
Traceback (most recent call last):
  File "button_event.py", line 49, in main
    GPIO.wait_for_edge(but_pin, GPIO.FALLING,timeout=1)
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 551, in wait_for_edge
    raise RuntimeError("Conflicting edge detection event already exists "
RuntimeError: Conflicting edge detection event already exists for this GPIO channel

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio_cdev.py", line 224, in close_line
    os.close(line_handle)
OSError: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "button_event.py", line 61, in <module>
    main()
  File "button_event.py", line 58, in main
    GPIO.cleanup()  # cleanup all GPIOs
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 381, in cleanup
    _cleanup_all()
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 273, in _cleanup_all
    _cleanup_one(ch_info)
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 263, in _cleanup_one
    gpio_cdev.close_line(ch_info.line_handle)
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio_cdev.py", line 226, in close_line
    raise GPIOError(e.errno, "Closing existing GPIO line: " + e.strerror)
Jetson.GPIO.gpio_cdev.GPIOError: [Errno 9] Closing existing GPIO line: Bad file descriptor

anhmiuhv, 2024-01-06:
Yeah you can do something like that. I think ideally, wait_for_edge should not have problem so I need to check it further. But you can use add_event_detect for now.

anhmiuhv, 2024-01-06:
Can you try the fix in #104 to see if it fixes the problem.

Romihi, 2024-01-14:
@anhmiuhv 
Hi, I tried to run the script, test_all_apis.py and got some errors on the the other functions and also test_wait_for_edge_timeout. Still some bugs there?
add_event_detect works fine though.

**For test_setup_twice, test_out_in_init_high, and test_out_in_init_low, I got the error like below:**
--
(base) jetson@jetson-desktop:~/projects/jetson-gpio/samples$ python test_all_apis.py
Testing test_warnings_off
Testing test_warnings_on
Testing test_setup_one_board
Testing test_setup_one_bcm
Testing test_setup_one_cvm
Testing test_setup_one_tegra_soc
Testing test_setup_twice
Traceback (most recent call last):
  File "test_all_apis.py", line 1084, in <module>
    test()
  File "test_all_apis.py", line 281, in test_setup_twice
    assert(val == GPIO.HIGH)
AssertionError

**For test_wait_for_edge_timeout(after commenting out the above functions):**
--
(base) jetson@jetson-desktop:~/projects/jetson-gpio/samples$ python test_all_apis.py
Testing test_warnings_off
Testing test_warnings_on
Testing test_setup_one_board
Testing test_setup_one_bcm
Testing test_setup_one_cvm
Testing test_setup_one_tegra_soc
Testing test_setup_one_out_no_init
Testing test_setup_one_out_high
Testing test_setup_one_out_low
Testing test_setup_many_out_no_init
Testing test_setup_many_out_one_init
Testing test_setup_many_out_many_init
Testing test_setup_one_in
Testing test_setup_one_in_pull
/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py:345: UserWarning: Jetson.GPIO ignores setup()'s pull_up_down parameter
  warnings.warn("Jetson.GPIO ignores setup()'s pull_up_down parameter")
Testing test_setup_many_in
Testing test_setup_all
Testing test_cleanup_one
Testing test_cleanup_many
Testing test_cleanup_all
Testing test_input
Testing test_output_one
Testing test_output_many_one_value
Testing test_output_many_many_value
Testing test_gpio_function_unexported
Testing test_gpio_function_in
Testing test_gpio_function_out
Testing test_wait_for_edge_timeout
Traceback (most recent call last):
  File "test_all_apis.py", line 1084, in <module>
    test()
  File "test_all_apis.py", line 510, in test_wait_for_edge_timeout
    val = GPIO.wait_for_edge(pin_data['in_a'], GPIO.BOTH, timeout=1)
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 551, in wait_for_edge
    raise RuntimeError("Conflicting edge detection event already exists "
RuntimeError: Conflicting edge detection event already exists for this GPIO channel

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio_cdev.py", line 224, in close_line
    os.close(line_handle)
OSError: [Errno 9] Bad file descriptor

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "test_all_apis.py", line 1088, in <module>
    GPIO.cleanup()
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 381, in cleanup
    _cleanup_all()
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 273, in _cleanup_all
    _cleanup_one(ch_info)
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio.py", line 263, in _cleanup_one
    gpio_cdev.close_line(ch_info.line_handle)
  File "/home/jetson/miniconda3/lib/python3.8/site-packages/Jetson/GPIO/gpio_cdev.py", line 226, in close_line
    raise GPIOError(e.errno, "Closing existing GPIO line: " + e.strerror)
Jetson.GPIO.gpio_cdev.GPIOError: [Errno 9] Closing existing GPIO line: Bad file descriptor

anhmiuhv, 2024-01-15:
Did you update Jetson GPIO?

Romihi, 2024-01-17:
@anhmiuhv 
Yes. I pulled your recent update.
https://github.com/NVIDIA/jetson-gpio/compare/37f1586904a8ccf41cabd0f3a1a5ee328c376e3b..f5f130ff9726b3ce8ed3363df01da76e52464304

"test_all_apis.py" has the below lines:
@test
def test_wait_for_edge_timeout():
    GPIO.setmode(GPIO.BOARD)
    GPIO.setup(pin_data['out_a'], GPIO.OUT, initial=GPIO.LOW)
    GPIO.setup(pin_data['in_a'], GPIO.IN)
    for i in range(3):
        val = GPIO.wait_for_edge(pin_data['in_a'], GPIO.BOTH, timeout=1)
        assert val is None
    GPIO.cleanup()
---
gpio_event.py has also updated.

anhmiuhv, 2024-01-17:
I think the update you pull is missing some changes in gpio_event.py. can you check #104 again or update Jetson GPIO to the latest version on pypi

anhmiuhv, 2024-01-17:
Please see this diff
https://github.com/NVIDIA/jetson-gpio/compare/37f1586904a8ccf41cabd0f3a1a5ee328c376e3b...f5f130ff9726b3ce8ed3363df01da76e52464304

Romihi, 2024-01-20:
@anhmiuhv 
Thank you for your following up.
It finally works after writing the diff into my code directly.
I do't know why the git  command `git pull origin master` did not pull your update and also `pip install Jetson.GPIO` did't work either.　Both pulled ver. 2.1.5...
But anyway thanks to fix it!

closed 2024-01-21 by anhmiuhv