NVIDIA/jetson-gpio #106: Jetson model not detected - Jetson Orin AGX 64GB Developer Kit

pbrandstet, 2024-01-26:
Hy.

When I do
>>> import Jetson.GPIO as GPIO

I get

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/__init__.py", line 1, in <module>
    from .gpio import *
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio.py", line 69, in <module>
    model, JETSON_INFO, _channel_data_by_mode = gpio_pin_data.get_data()
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 594, in get_data
    model = get_model()
  File "/usr/lib/python3/dist-packages/Jetson/GPIO/gpio_pin_data.py", line 590, in get_model
    raise Exception('Could not determine Jetson model')
Exception: Could not determine Jetson model

How can i fix that?

Thanks

pbrandstet, 2024-01-27:
I fixed it.
Would be nice if you add the information that a environment variable needs to be set.
I only realized it by debugging the libary.

closed 2024-02-06 by anhmiuhv

dingkwang, 2024-07-24:
Hi @pbrandstet , which environment variable  did you set?

pbrandstet, 2024-07-24:
lib/python/Jetson/GPIO/gpio_pin_data.py

Check line 35 and 593.

JETSON_MODEL_NAME should be the correct env var.

MAGICXQY, 2024-11-18:
so  How to modify this Python so that AGX ORIN can recognize it