NVIDIA/jetson-gpio #108: Multi-Pin

Satge96, 2024-03-20:
I wanted to ask how it would be possible to change multiple pins truly simultaneously. Can I do that in Python?

anhmiuhv, 2024-03-20:
Is this something you want to do? https://stackoverflow.com/questions/11523918/start-a-function-at-given-time

However, the GPIO bank in the Jetson would process the request sequentially due to hardware limitation. there will be a very small time delay between the pins.

Satge96, 2024-03-20:
The time when the Pins are toggled is not relevant to me as long as it's at the same time. Would you have an idea or even an example for that?

anhmiuhv, 2024-03-20:
[def output(channels, values):](https://github.com/NVIDIA/jetson-gpio/blob/2b6da70499175f97a581423c15b20f34d4dd43b4/lib/python/Jetson/GPIO/gpio.py#L407)

output can take a list of GPIO and a list of corresponding values.

closed 2024-07-28 by anhmiuhv