NVIDIA/jetson-gpio #110: Jetapck 6 Orin Nano/NX Pin 32 Not Working

AlexKlimaj, 2024-05-15:
Coming from everything working in Jetpack 5, I am unable to get pin 32 working in Jetpack 6. Pin 33 works, Pin 32 doesn't. It does work through gpioset but not through jetson-gpio.

https://forums.developer.nvidia.com/t/gpio07-pg-06-not-working-in-jetpack-6/293013

AlexKlimaj, 2024-05-17:
Okay it is actually working but the pin state reverts to default when the script exits. In Jetpack 5 the pin stays in the state its in when the script exits. In Jetpack 6, the pin reverts back to the default pinmux state.

anhmiuhv, 2024-05-17:
we clean up the pin state when python code exit.
This is what I found for gpioset. We dont support different mode like this for the library.

Modes:[¶](https://manpages.debian.org/testing/gpiod/gpioset.1.en.html#Modes:)
[exit:](https://manpages.debian.org/testing/gpiod/gpioset.1.en.html#exit:)
set values and exit immediately
[wait:](https://manpages.debian.org/testing/gpiod/gpioset.1.en.html#wait:)
set values and wait for user to press ENTER
[time:](https://manpages.debian.org/testing/gpiod/gpioset.1.en.html#time:)
set values and sleep for a specified amount of time
[signal:](https://manpages.debian.org/testing/gpiod/gpioset.1.en.html#signal:)
set values and wait for SIGINT or SIGTERM
Note: the state of a GPIO line controlled over the character device reverts to default when the last process referencing the file descriptor representing the device file exits. This means that it's wrong to run gpioset, have it exit and expect the line to continue being driven high or low. It may happen if given pin is floating but it must be interpreted as undefined behavior.

anhmiuhv, 2024-05-17:
we clean up the pin state when python code exit. We dont support different mode like this for the library.

AlexKlimaj, 2024-05-17:
I'm not concerned with the behavior of gpioset but this python library. If you don't call cleanup, it should leave the pin in the state you set it to.

AlexKlimaj, 2024-05-18:
Okay so this comes back to the fundamental change over to libgpiod.

I changed my pinmux on this pin to output drive high by default. But even in the python script that only ever drives the pin high, when the script exits, it goes back to low.

Any interaction with the pin will cause it to reset to low.

aliakbar-cobot, 2024-11-12:
Hi, has there been any update to this ticket? I'd like to be able to set a GPIO pin to high and have it stay high after the python script exits.

anhmiuhv, 2024-11-12:
Hi this will require a kernel driver change. Please set the pinmux on your pin to output drive high by default and then apply the patch in here

https://forums.developer.nvidia.com/t/40hdr-spi1-gpio-padctl-register-bit-10-effect-by-gpiod-tools-in-jp6/301171/19

closed 2024-11-12 by anhmiuhv