NVIDIA/jetson-gpio #111: Remove Python 2 requirement

paulober, 2024-07-27:
**Please don't submit questions**
Please use this form to report specific bugs in Jetson.GPIO. All general questions or support requests should be directed at NVIDIA's forums; see:
https://developer.nvidia.com/embedded/community/support-resources

Please delete this section from your bug report.

**Describe the bug**
A clear and concise description of what the bug is, including
- All newest version as of 28.07.2024
- On Jetson Orin NX 16GB

**To Reproduce**
Steps to reproduce the behavior:
1. Setup new device

**Expected behavior**
The only package dependent on python2 is this module. As python2 is way out of it's service period and contains many security vulnerabilities it would be great if this module could dump it from the dependencies list, so L4T can be Python2 less setup.

**Additional context**
Add any other context about the problem here.

anhmiuhv, 2024-07-28:
I think this module has separate packages for python3 and python2. So all you need to do is uninstall the package for python2

paulober, 2024-07-28:
> I think this module has separate packages for python3 and python2. So all you need to do is uninstall the package for python2

That would be great but uninstalling python2 will show following packages to be removed: `python-jetson-gpio python2 python2.7`

anhmiuhv, 2024-07-28:
The OS does not rely on any packages from Jetson GPIO to run

paulober, 2024-07-28:
> The OS does not rely on any packages from Jetson GPIO to run
ok, thanks.

closed 2024-07-28 by paulober