NVIDIA/jetson-gpio #113: how to access pins in gpiochip1 for orin nano

arrow53, 2024-09-03:
I've modified `gpio_pin_data.py` to add two new outputs `GPIO3_PP.06` and `GPIO3_PCC.00`. 

The line offset for `PP.06` is easy as that's in `gpiochip0` so that works fine. The `PCC.00` one though is in `gpiochip1` and that doesn't work. I figured the offset would just be 164 + 12 but no joy. 

Any ideas?
```
JETSON_ORIN_NX_PIN_DEFS = [                                                     
    (98, 'PP.06', "tegra234-gpio", 7, 4, 'GPIO02', 'GP02', None, None),         
    (176, 'PCC.00', "tegra234-gpio", 7, 4, 'GPIO03', 'GP03', None, None),       
    (144, 'PAC.06', "tegra234-gpio", 7, 4, 'GPIO09', 'GP167', None, None),   
```

anhmiuhv, 2024-09-05:
can you try to change tegra234-gpio to tegra234-gpio-aon

anhmiuhv, 2024-09-16:
If all is good i will close

closed 2024-09-16 by anhmiuhv