NVIDIA/jetson-gpio #114: jetson-gpio doesn't work on Jetpack 6.0 ???

HappySamuel, 2024-09-12:
Recently, i upgrade my AGX Orin from Jetpack 5.0 to 6.0, but jetson-gpio seems doesn't work anymore. Any idea why? and how to solve this issue?

anhmiuhv, 2024-09-12:
How did you upgrade it?

HappySamuel, 2024-09-13:
Hi @anhmiuhv 

Flashing Jetpack 6.0 into AGX Orin

roshea6, 2024-09-16:
I seem to be getting the same issue on a Jetson Orin NX development kit flashed with Jetpack 6.0. I've used the library without issue before on a Jetson Xavier NX on Jetpack 5.

anhmiuhv, 2024-09-16:
In Jetpack 6.0 we use the upstream kernel gpio driver instead of the downstream kernel driver in JP5. The upstream driver does not have the functionality to dynamically change the GPIO into gpio mode. Therefore, you will need to modify the pinmux in the BCT to configure the pin you want to use as GPIO and flash it onto the device. 

This is the documentation on how to modify and create the BCT
https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonAgxOrinSeries.html?highlight=pin%20direction#pinmux-changes

To use a pin as GPIO, ensure that E_IO_HV field is disabled in corresponding pinmux register of the GPIO pin. You can disable the 3.3V Tolerance Enable field in the pinmux spreadsheet.
Also, make sure Pin Direction to Bidirectional so that userspace framework can operate GPIO in both input and output direction. After these configurations, reflash the board with the updated pinmux file

SamRodkey, 2024-10-25:
> In Jetpack 6.0 we use the upstream kernel gpio driver instead of the downstream kernel driver in JP5. The upstream driver does not have the functionality to dynamically change the GPIO into gpio [...]
This statement makes it sound like GPIO functionality (and the Jetson-GPIO library) is broken for unmodified Jetson Orin Nano and other units which can use Jetpack 6 without performing these steps. Is this correct?

> This is the documentation on how to modify and create the BCT https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonAgxOrinSeries.html?highlight=pin%20direction#pinmux-changes
> 
Is the link here essentially the required steps to restore GPIO functionality when using Jetpack 6 for any NVIDIA Jetson series using Jetpack 6 other than a Jetson Orin AGX Developer Kit board?

The following statement is ambiguous to me:
>If your board schematic differs from the schematic for the Jetson AGX Orin Developer Kit board, you must change the pinmux configuration applied by the software.
I'm not sure if this statement is saying any stock, unmodified board purchased by NVIDIA other than AGX Orin Developer Kits or AGX Orin developer kits which have a modified physical interface and therefore would have a different schematic...

> To use a pin as GPIO, ensure that E_IO_HV field is disabled in corresponding pinmux register of the GPIO pin. You can disable the 3.3V Tolerance Enable field in the pinmux spreadsheet. Also, make [...]
Can you elaborate on this. What does the `E_IO_HV` field do? I only use a python wrapper of this library, I have no idea what this means....

SamRodkey, 2024-10-25:
> This is the documentation on how to modify and create the BCT https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonAgxOrinSeries.html?highlight=pin%20direction#pinmux-changes

I ran the spreadsheet and the macro per the instructions using Excel on a MacOS and it appears to have failed with no warnings or errors. The only output is a zero byte file in the parent directory where the directory path is included in the file name?

The macro is located inside `/downloads/pinmux/` and it created a zero byte file called `pinmux\orin-jetson orin nano-pinmux.dtsi`. I think macro only works on a windows machine due to path separator differences between platforms?

jonas-la, 2025-01-23:
Once you have the updated pinmux.dtsi how to you configure flash.sh to use this file for the pinmux configuration?

anhmiuhv, 2025-02-14:
You can do this https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonAgxOrinSeries.html?highlight=pin%20direction#pinmux-changes

See the notes section

> Copy the pinmux.dtsi file to the <l4t_top>/bootloader/generic/BCT/ directory, and copy the gpio.dtsi file to the <l4t_top>/bootloader/ directory.

> You do not need to copy the padvoltage.dtsi file.

anhmiuhv, 2025-02-14:
Note that you can dynamically change the pinmux also. That way you can change GPIO pinmux without having to reflash the mb1-bct
https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonAgxOrinSeries.html?highlight=pin%20direction#changing-the-pinmux

alexanderywy, 2025-04-10:
Hello @anhmiuhv , I saw the last commit is in Feb 8, 2025. Have these problems in JetPack 6.0 solved? I found my Jetson.GPIO is not working properly too. I checked my Jetson.GPIO's version is 2.1.7.
Is that feasible to set pins by using the lateset 2.1.9? Or it's still necessay to modify stuff like pinmux?

alexanderywy, 2025-04-10:
And if 2.1.9 solved these problem, I wonder how to upgrade to 2.1.9 ?

alexanderywy, 2025-04-10:
> You still have to do this for now. You can update through apt update or pip (https://github.com/NVIDIA/jetson-gpio?tab=readme-ov-file#using-pip)

Ok，I will try to set GPIO manually. Thanks!Will this issue be fixed in future ?发自我的 iPhone

anhmiuhv, 2025-04-10:
I cannot confirm anything in the roadmap but this is something we want to improve upon in the future.