NVIDIA/jetson-gpio #115: Carrier board support warning message

shkipan, 2024-11-11:
Hello!
I have installed the library using pip following the readme steps smoothly. However, the script throws the following warning:
>WARNING: Carrier board is not from a Jetson Developer Kit.
>WARNNIG: Jetson.GPIO library has not been verified with this carrier board,
>WARNING: and in fact is unlikely to work correctly.
 
And the pins control commands seems to fail (the multimeter shows the low voltage on the pin which is supposed to have high voltage on it.

Here's the debug info which python has returned to me.
>GPIO.JETSON_INFO
{'P1_REVISION': 1, 'RAM': '32768M, 65536M', 'REVISION': 'Unknown', 'TYPE': 'JETSON_ORIN_NX', 'MANUFACTURER': 'NVIDIA', 'PROCESSOR': 'A78AE'}

 I am using Orin NX with the official devkit board, the Jetpack version 6. 

Is this a bug or am I doing something wrong?

anhmiuhv, 2024-11-11:
can you check #114 ?

anhmiuhv, 2024-11-12:
also can you run 

tree /proc/device-tree/chosen/ids

anhmiuhv, 2025-02-06:
Close as no response for a long time

closed 2025-02-06 by anhmiuhv