NVIDIA/jetson-gpio #116: When using jetpack 6.1 with jetson-gpio, the message "Could not determine Jetson model" appears?

EESN-W, 2024-12-20:
I upgraded the Jetson Orin Nano Developer Kit to Super (jetpack 6.1). How long will it take to get the GPIO control function?

EESN-W, 2024-12-20:
[image]

EESN-W, 2024-12-20:
[image]

meetwudi, 2024-12-22:
`export JETSON_MODEL_NAME=JETSON_NANO` temporarily works around this. Turns out this environment variable is no longer set in 6.1 rev.1.

EESN-W, 2024-12-23:
> `export JETSON_MODEL_NAME=JETSON_NANO` temporarily works around this. Turns out this environment variable is no longer set in 6.1 rev.1.

Thank you for your reply. I can successfully read the pin level by setting export JETSON_MODEL_NAME=JETSON_ORIN_NANO.

dakejahl, 2025-01-30:
Tested on a Jetson Orin Nano with Jetpack 6.2 (r36.4.3) and the issue still persists. Setting this solves it
```
export JETSON_MODEL_NAME=JETSON_ORIN_NANO
```

anhmiuhv, 2025-02-06:
The latest Jetson.GPIO version 2.1.8 should be supporting Jetson Orin Nano Super. If you still have a problem, please refer to this
https://github.com/NVIDIA/jetson-gpio/blob/fe01769407cf1359786a561d40251aecd1aa6d92/README.md?plain=1#L453
for how the Jetson.GPIO detects the hardware it is running on. 

Setting the environment variable like this is a way to support unsupported hardware
`export JETSON_MODEL_NAME=JETSON_ORIN_NANO
`

dakejahl, 2025-02-06:
@anhmiuhv I just tried 2.1.8 and it's the same issue. Looking at /proc/device-tree I see that my board is correctly identified as a Jetson Orin Nano Super

anhmiuhv, 2025-02-07:
Let me take a look again

anhmiuhv, 2025-02-07:
> [@anhmiuhv](https://github.com/anhmiuhv) I just tried 2.1.8 and it's the same issue. Looking at /proc/device-tree I see that my board is correctly identified as a Jetson Orin Nano Super
> 
> ```
> jetson@jetson:~$ cat /proc/device-tree/compatible
> nvidia,p3768-0000+p3767-0004-supernvidia,p3767-0004nvidia,tegra234
> ```

Can you take a look again?

closed 2025-02-14 by anhmiuhv