NVIDIA/jetson-gpio #118: How to do that it is available to use Rogue Carrier on your GPIO library?

st-suzuki-sec, 2025-01-15:
Hello!
We use Jetson AGX Orin Industrial and Rogue Carrier.
How to do that it is available to use Rogue Carrier on your GPIO library. Is it impossible?
Do we need to add definition of its Carrier in files like gpio_pin_data.py under jetson-gpio/lib/python/Jetson/GPIO/ directory?

If you have idea, please tell us the correct way to fix.

Thanks

anhmiuhv, 2025-01-15:
https://github.com/NVIDIA/jetson-gpio/commit/3200bc5d4ceac7a5f80c20fa3dd77363cfd6ed51

Yes that is correct. You can use this commit as a reference. This commit add support for Orin Nano, which has similar PIN layout as Orin NX. If you Carrier has different pin layouts comparing to the AGX Orin than you will need to define a new Pin layout also

st-suzuki-sec, 2025-01-15:
Thank you for your prompt response!
I will check its reference and try it!

st-suzuki-sec, 2025-01-17:
Hi!
I checked above-mentioned reference.
I'm not sure how to define *_PIN_DEFS array.
Could
https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L324-L349

https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L372

beginner

st-suzuki-sec, 2025-01-17:
Hi!
I checked above-mentioned commit reference.
I'm not sure how to define *_PIN_DEFS array(*1) and RAM(*2) setting.
I investigated below carrier for Orin Nano, but  I couldn't find the same definition between `JETSON_NANO_PIN_DEFS` and below Pin layout.
So could you tell me more detail how to define and Where Pin layout document is?? Cause I'm beginner.

Thanks

(*1)
https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L324-L349

(*2)
https://github.com/NVIDIA/jetson-gpio/blob/master/lib/python/Jetson/GPIO/gpio_pin_data.py#L372
```
    JETSON_ORIN_NANO: (
        JETSON_ORIN_NX_PIN_DEFS,
        {
            'P1_REVISION': 1,
            'RAM': '32768M, 65536M', # ← here
            'REVISION': 'Unknown',
            'TYPE': 'JETSON_ORIN_NANO',
            'MANUFACTURER': 'NVIDIA',
            'PROCESSOR': 'A78AE'
        }
    ),
```

anhmiuhv, 2025-02-06:
Jetson Orin Nano use the same Pin dev as JETSON_ORIN_NX_PIN_DEFS

This part is just for information. It serves no functional purpose so you can ignore it

        {
            'P1_REVISION': 1,
            'RAM': '32768M, 65536M', # ← here
            'REVISION': 'Unknown',
            'TYPE': 'JETSON_ORIN_NANO',
            'MANUFACTURER': 'NVIDIA',
            'PROCESSOR': 'A78AE'
        }