NVIDIA/jetson-gpio #120: GPIO not working on Jetson Orin Nano Super Developer Kit

sarmadgulzar, 2025-02-13:
Device: NVIDIA Jetson Orin Nano Super Developer Kit
OS: Ubuntu 22.04.5 LTS (Jammy Jellyfish) with Jetpack 6.2
SSD: Samsung 980 NVMe M.2

The LED is not blinking when I run the `samples/simple_out.py`

Here's my code:
```
$ cat blink_led.py 
import RPi.GPIO as GPIO
import time

# Pin Definitions
output_pin = 18  # BCM pin 18, BOARD pin 12

def main():
    # Pin Setup:
    GPIO.setmode(GPIO.BCM)  # BCM pin-numbering scheme from Raspberry Pi
    # set pin as an output pin with optional initial state of HIGH
    GPIO.setup(output_pin, GPIO.OUT, initial=GPIO.HIGH)

    print("Starting demo now! Press CTRL+C to exit")
    curr_value = GPIO.HIGH
    try:
        while True:
            time.sleep(1)
            # Toggle the output every second
            print("Outputting {} to pin {}".format(curr_value, output_pin))
            GPIO.output(output_pin, curr_value)
            curr_value ^= GPIO.HIGH
    finally:
        GPIO.cleanup()

if __name__ == '__main__':
    main()
```

I've installed the library into the global Python installation:
```
$ pip3 freeze | grep Jetson
Jetson.GPIO==2.1.9
```

I've setup the permissions as described here: https://github.com/NVIDIA/jetson-gpio?tab=readme-ov-file#setting-user-permissions

I've also run `export JETSON_MODEL_NAME=JETSON_ORIN_NANO` as described in #116 

The positive leg of LED is connected to pin `12` on the board and the negative leg is connected to GND on pin `6`. I can confirm that LED works if I connect the positive log of the LED to pin `1` which is `3.3v`.

Any ideas what I might be doing wrong?

anhmiuhv, 2025-02-14:
Please refer to
https://github.com/NVIDIA/jetson-gpio/issues/114#issuecomment-2353766759

anhmiuhv, 2025-02-14:
You will need to change the pinmux for Jetson Orin Nano

https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonOrinNxNanoSeries.html#generating-the-pinmux-dtsi-files

sarmadgulzar, 2025-02-21:
@anhmiuhv thanks for the reference to the documentation. I was able to dynamically change the registers and make the LED blink. I'll post more on that later so that others can use it but before that I have a question. How do we determine the mapping between the pins and the registers? Let's say I want to use the following pin (Board pin # 31):

[image]

This corresponds to this register `PADCTL_G3_SOC_GPIO33_0`: 

[image]

How exactly do we determine that? Although I was able to run the LED but that's only because I knew this pin <-> register mapping from the example given here: https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonOrinNxNanoSeries.html#changing-the-pinmux

If you can guide me in the right direction, that'll be super helpful. Thanks again!

anhmiuhv, 2025-02-21:
Step 1: Check the carrier board spec sheet.

[image]

As you can see here, you want to use Pin 31, which corresponds with PQ.06

Step 2: Open the corresponding pinmux spreadsheet for your device type. In this example, you use Jetson Orin Nano so we are going to use Jetson_Orin_NX_series_and_Orin_Nano_series_Pinmux_Config_Template.xlsm . You can find this file on the Jetson download center.

Step 3: Find PQ.06 in the pinmux spreadsheet.

[image]

You can see that in here PQ.06 and SOC_GPIO33 are in the same row

Step 4: Find the corresponding PADCTL register for SOC_GPIO33 in the Orin Technical Reference Manual like what you do here:

[image]

sarmadgulzar, 2025-02-21:
@anhmiuhv thanks. Much appreciated!

aryans99860, 2025-03-03:
@sarmadgulzar Hey! Can you share the step by step way to set GPIO pin bidirectional/Output. I am facing problems doing this. 
Thanks

alexanderywy, 2025-04-11:
> You will need to change the pinmux for Jetson Orin Nano
> 
> https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonOrinNxNanoSeries.html#generating-the-pinmux-dtsi-files
> 
> From the docs:
> 
> > Starting with JetPack 6, you can change the pinmux in the following ways:
> > Update the MB1 pinmux BCT as mentioned in [Generating the Pinmux dtsi Files]?(https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonOrinNxNanoSeries.html#generating-the-pinmux-dtsi-files).
> > For debugging, you can dynamically change the pinmux.
> 
> I suggest you dynamically change the pinmux first to make the GPIO output work first

Sorry for bothering, I am now using Jetpack 6.0 with Jetson.GPIO == 2.1.7. Is that means dynamically change is to use jetson-io.py, so that I can use setup() in Jetson.GPIO? Or maybe I misunderstood what is dynamically change.

alexanderywy, 2025-04-16:
> Dynamically change the pinmux mean using busybox devmem tool
> 
> ```
> busybox devmem
> ```
> 
> You can read more about it in the documentation here https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonOrinNxNanoSeries.html#generating-the-pinmux-dtsi-files

So in JETPACK 6.x, after setup pins manually, can we still use Jetson.GPIO to enable functions? Does it only affect GPIO.setup()?

anhmiuhv, 2025-04-16:
Yes you can use Jetson GPIO to use GPIO after changing the pinmux

vrano714, 2025-04-25:
@anhmiuhv @sarmadgulzar Thank you for amazing step-by-step description! I faced the same problem (Jetson.GPIO cannot set HIGH/LOW of specific GPIO pin) with Jetson AGX Orin (DP6.2) and could solved it with that description.

Actually Jetson AGX Orin seems to have different pin configurations, I had to look the documents (TRM, Board specification, pinmux excel) related to AGX Orin.

Shivang-10x, 2025-05-30:
Hello.
I am facing the same issue as @vrano714 , but am unable to fix it.
I have a Jetson Orin AGX with Jetpack 6.2 installed using the SDK manager GUI on an Ubuntu 22.04 system. I want to toggle a GPIO pin every 1 second (board pin 7/ SOC_GPIO33 eg from https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonOrinNxNanoSeries.html#changing-the-pinmux).

I have attempted to set the state of the pinmux dynamically as shown in the example with:
`sudo busybox devmem 0x02430070 w 0x400`
Re-reading the register gives: `0x00000400`

I am now attempting to toggle the pins using the Jetson-GPIO library using the following code:
`import Jetson.GPIO as GPIO
import time

GPIO_PIN = 7

GPIO.setmode(GPIO.BOARD)
GPIO.setup(GPIO_PIN, GPIO.OUT, initial=GPIO.LOW)

for _ in range(5):
    GPIO.output(GPIO_PIN, GPIO.HIGH)
    time.sleep(1)
    GPIO.output(GPIO_PIN, GPIO.LOW)
    time.sleep(1)

GPIO.cleanup()`

When running the above code with and without `sudo` , using a multimeter to read the potential, I am getting voltage toggling between from 1.594V - 1.577V which does seem to toggle every second.

What am I missing ?

hrege24, 2025-10-07:
Hi @anhmiuhv , thanks for the detailed instructions here, this is very helpful! I followed the steps to dynamically change the pinmux from input to output, however, I'm still unable to get the GPIO to properly turn on an LED. I confirmed that the pinmux was properly updated, and the built-in warnings were cleared:

`>>> GPIO.setup([LED1_RED, LED1_GREEN], GPIO.OUT)
/usr/lib/python3.12/site-packages/Jetson/GPIO/gpio_cdev.py:361: UserWarning:
[WARNING] User requested output for channel "32", but it is set to input in pinmux. For more information on resolving this, please see
https://docs.nvidia.com/jetson/archives/r36.3/DeveloperGuide/HR/JetsonModuleAdaptationAndBringUp/JetsonOrinNxNanoSeries.html#generating-the-pinmux-dtsi-files

This can be resolved *temporarily* (until next restart) by running:
    sudo busybox devmem 0x2434080 w 0x5

  warnings.warn(
>>> addr = 0x2434080
>>> value = 0x5
>>> fd = os.open("/dev/mem", os.O_RDWR | os.O_SYNC)
>>> page_size = mmap.PAGESIZE
>>> page = addr & ~(page_size - 1)
>>>
>>> mem = mmap.mmap(fd, page_size, mmap.MAP_SHARED, mmap.PROT_WRITE | mmap.PROT_READ, offset=page)
>>> offset = addr - page
>>> mem[offset:offset+4] = struct.pack("I", value)
>>> mem.close()
>>> os.close(fd)
>>> GPIO.setup([LED1_RED, LED1_GREEN], GPIO.OUT)`

What should the `value` be to enable 3.3V or higher for the GPIO output? I tried 0x004 as well, with no success.

husain-zaidi, 2026-02-15:
updating to Jetpack 6.2.2 helps as now jetson-io.py has option to enable gpio for pins individually

[image]